spectraphilic/wsn_riot #1: Compatibility

ArcticSnow, 2020-05-06:
**Natively supported by RIOT that we currently use:**
- DS18b20
- BME280
- DS1307 (RTC)
-xbee (not completely sure to which extent is compatible with digimesh)
- lora (compatible with libelium chip?)
- protocols:
  - i2c
  - UART
  - SPI  (sd card)

**No native support**:
- MLX
- VL
- AS7265X
- DS2 / CTD10   (SDI-12 protocol not natively supported)
- SHT31 (in development apparently)
- TMP
- waspmote v15 (with arduino compiling)

Guidelines for evaluation/review:
- support for waspmote v15?
- usage of arduino code?
- foreseen hurdles vs future capabilities? What edge in comparison to Arduino?

ArcticSnow, 2020-07-27:
**Accomplished**
- flash code to Wasp v15
- BME280 and SHT31 confirmed to work. 
- started testing DS18b20 (code)

**TODO**
- read and write to SD
- send packet via radio (check if Re-mote radio freq is compatible with Libelium Lora module.
- figure out RTC hardware/software for waspmote (in progress)
- find out best path forward to run not currently supported sensors. Should we start native libraries?
- Once all the steps above are validated, we can start implementing the logger program in full mode.

jdavid, 2020-08-13:
With the latest commit now the **test-shel**l app mounts the SD card and there're new `ls` and `vfs` commands that work, tested with the Re-mote.

The SD card uses SPI. Above that there's the MTD layer, where MTD stands for Memory Technology Device, my understanding is that this is a hardware abstraction, idea borrowed from Linux as many other things in RIOT.

Then we've the filesystem, we use FAT but RIOT supports some more. And above the filesystem there is the VFS layer, where VFS stands for Virtual File System. Using VFS we should be able to switch to a different filesystem without changing user's code.
Probably in the future we should use littlefs instead of fatfs (low priority though).

jdavid, 2020-08-13:
In our current stack we use the EEPROM to store configuration, but apparently the re-mote doesn't have EEPROM. So we will use the SD card instead.

jdavid, 2020-10-05:
Some RIOT projects to track:

- New drivers https://github.com/RIOT-OS/RIOT/projects/29
- CC2538 (Remote's MCU) https://github.com/RIOT-OS/RIOT/projects/5

closed 2021-11-23 by ArcticSnow