spectraphilic/wsn_riot #10: Arduino drivers

jdavid, 2020-10-15:
Many drivers we need are not natively supported by RIOT. One option is to use an Arduino driver. Apparently this is possible, see https://github.com/RIOT-OS/RIOT/pull/12518

But first we need our boards to provide the arduino feature, and this is not supported by any of our boards, for example:

```
$ CONTINUE_ON_EXPECTED_ERRORS=1 BOARD=waspmote-pro make -C tests/sys_arduino
make: Entering directory '/home/jdavid/sandboxes/UiO/wsn_riot/RIOT/tests/sys_arduino'
There are unsatisfied feature requirements: arduino
[...]
In file included from /home/jdavid/sandboxes/UiO/wsn_riot/RIOT/sys/arduino/base.cpp:28:
/home/jdavid/sandboxes/UiO/wsn_riot/RIOT/sys/arduino/include/arduino.hpp:27:10: fatal error: arduino_board.h: No such file or directory
   27 | #include "arduino_board.h"
[...]
```

Looking for `arduino_board.h` we see there are already a number of boards that support the arduino feature, including some with a non-AVR MCU. But none of our boards does: waspmote-pro, remote-revb and feather-m0.

Also, from what I've read the **arduino** module allows to run unmodified Arduino sketches. While the **arduino_lib** pseudomodule allows to use Arduino libraries from RIOT applications, and that's exactly what we want. See https://github.com/RIOT-OS/RIOT/pull/12180

In order to validate this approach we need to:

1. [ ] Add at least arduino_board.h to one of our boards, for example to waspmote-pro
2. [ ] Get at least `tests/sys_arduino` and `tests/sys_arduino_lib` working
3. [ ] Add a driver as a package and verify it works

jdavid, 2020-10-15:
See `boards/common/arduino-atmega/include/arduino_board.h` for an example.

jdavid, 2020-10-15:
Follow https://github.com/RIOT-OS/RIOT/issues/15232

jdavid, 2020-10-21:
Add Arduino support to a board
https://doc.riot-os.org/group__sys__arduino.html#sec_boardsupport

ArcticSnow, 2020-10-27:
Arduino drivers for the Feather boards in ArduinoIDE can be found here: https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/WVariant.h#L110

jdavid, 2020-10-27:
I've packaged Sparkfun's BME280 Arduino library and added a test app, see `apps/test-arduino`. And tested it with Arduino Uno.
The main purpose of this issue was to evaluate whether we could use or not Arduino libraries, and that's done.

There's one caveat though, RIOT doesn't support packages out of their tree. So to test it I did a symbolic link from their tree to ours, such as:

```
$ cd RIOT/pkg
$ ln -s ../../pkg/arduino_sparkfun_bme280
$ cd ../..
$ BOARD=arduino-uno make -C apps/test-arduino/ flash term
```

I've opened an issue to fix this, see https://github.com/RIOT-OS/RIOT/issues/15309

I'm going to update the check list in the description of this issue.

jdavid, 2020-11-04:
I've added the arduino feature to the feather-m0 board in our RIOT fork, in the `feather-m0-arduino` branch, see https://github.com/spectraphilic/RIOT/tree/feather-m0-arduino

But I cannot submit a PR to RIOT upstream because `Serial.print` doesn't work. From what I've read we need first the following PR to be merged (pending confirmation):

- https://github.com/RIOT-OS/RIOT/pull/12268

jdavid, 2020-12-18:
I've makde a PR for the arduino feature in the remote-revb, https://github.com/RIOT-OS/RIOT/pull/15678

By the way there's already an I2C device in the remote-revb, the RTC at address 0x69, see https://github.com/spectraphilic/wsn_riot/issues/8

jdavid, 2022-10-31:
This can be closed. Latest board with Arduino support is lora-e5-dev
We also have support for SDI-12 through an Arduino library.

closed 2022-10-31 by jdavid